ketai/ketai #10: New version of PApplet is based on Fragment instead of Activity...

darkwave, 2015-08-22:
This new approach taken by the Processing team (that I personally appreciate) force the library source to be updated in order to reflect this change.

For example in many point of the source code...
<code>parent.getApplicationContext()</code>
must be changed into...
<code>parent.getActivity().getApplicationContext()</code>

I hope this info helps :sweat_smile:

totalgee, 2015-08-26:
Related to this -- I just installed Processing 3.0b5 and Ketai, and am trying a simple example with gyro sensor (this [example](http://ketai.org/examples/gyroscope/) from the website).  I get this crash when running -- caused by this problem.  The call to `new KetaiSensor(this)` fails with: `No virtual method getApplicationContext()`.  What is the workaround I can use in the meantime? Do I need to use an older Android SDK (I'm using 5.0.1)...?

ketai, 2015-08-26:
fixed in master...ongoing testing.

closed 2015-08-26 by ketai