ketai/ketai #9: Library not found after installing it (Android Mode 0241 and IDE 0243)

darkwave, 2015-08-22:
I updated Processing IDE to 0243 release and Android Mode to 0241 release and after installing Ketai Library using Library dialog it is not found by the environment and as a result it doesn't compile.

ketai, 2015-08-26:
any updates?  We were unable to reproduce.

ciaron, 2015-10-16:
I have the same problem. These are the steps I took:

0) confirm no Ketai/ folder exists in Processing libraries folder
1) From the Processing IDE, open the Contributions Manager (Tools -> Add Tool)
2) Select Ketai and install (v12 indicated under install button) - confirm Ketai/ folder now exists under libraries/
3) open a simple Ketai sketch (e.g. just "import ketai.camera.*")
4) sketch fails to build:
```
1. ERROR in /tmp/android3242095741886397564sketch/src/processing/test/sketch_151016b/sketch_151016b.java (at line 8)
	import ketai.camera.*; 
	       ^^^^^
The import ketai cannot be resolved
----------
1 problem (1 error)
```

Closing and re-opening Processing and opening the Contributions Manager shows Ketai as not installed - i.e. the green check-mark has disappeared, so something seems weird. Also: there are no examples under Examples -> Contributed Libraries, though they exist in the library folder.

However, a workaround exists: if I manually copy ketai.jar (or use "Add File...") into the sketch's code/ folder, Ketai works.

Processing 3.0, Android Mode 3.0-beta3

linux-man, 2015-11-08:
You can fix this issue if you lowercase the Kedai folder on sketchbook to kedai.
This bug is probably only noticed on Linux and Mac.

xuv, 2016-08-01:
I can confirm this bug and the solution by @linux-man.
Using Processing 3.1.1 on `Linux 4.6.4-1-ARCH #1 SMP PREEMPT Mon Jul 11 19:12:32 CEST 2016 x86_64 GNU/Linux`

AdoHaha, 2016-12-01:
same for me, working on Ubuntu 16.04 and Processing 3.2.3, although I had to change the folder name as well as Ketai to ketai in library.properties file. Please fix

ketai, 2017-09-06:
Properties file updated to match filename case. Tested on OS X, Linux (Kali/Ubuntu)

closed 2017-09-06 by ketai