ketai/ketai #11: android.content.Context Cannot be Resolved

DavidB-TPW, 2015-08-25:
I am trying to use Ketai to interact with a SQLite database, but I cannot because I keep getting two errors in the console:

`The type android.content.Context cannot be resolved. It is indirectly referenced from required .class files	The type android.content.Context cannot be resolved. It is indirectly referenced from required .class files`

`The method load(Context, String, String) from the type KetaiSQLite refers to the missing type Context	The method load(Context, String, String) from the type KetaiSQLite refers to the missing type Context`

Processing Version: `3.0 beta 5`
Processing Android Mode Version: `0242`

[image]

ketai, 2015-08-26:
fixed, based off of the PApplet moving to a Fragment vs. Activity previously

closed 2015-08-26 by ketai

DavidB-TPW, 2015-08-28:
I apologize, as I was not aware that a duplicate of this existed.