ketai/ketai #13: Support faster sensor update rate

totalgee, 2015-08-31:
It would be helpful to be able to specify the refresh rate for sensors (gyro, etc).  Currently, all registerListener() calls in KetaiSensor use `SensorManager.SENSOR_DELAY_UI`, but it would be nice to have an option to choose `SensorManager.SENSOR_DELAY_GAME` (or the other `SENSOR_DELAY_*` values).  The details of these seems to vary from system to system, but on my Moto G, `_GAME` gives much faster updates (about 40 per second) compared to `_UI`(which gives about 15 per second).

jkpoyzer, 2015-10-19:
Related comment:  I have been trying to up the GPS sensor update rate.  There seems to be a location.setUpdateRate( int millis, int meters) but I haven't been able to change it from the standard 10 second update.
	int refresh_rate = 1000;
	int refresh_dist = 10;

in Setup:
		location = new KetaiLocation(this);
		sensor = new KetaiSensor(this);
		sensor.start();
		location.setUpdateRate(refresh_rate, refresh_dist);

ketai, 2017-08-31:
added setSamplingRate(int pSamplingInterval) method to allow changing of sampling rate for sensors.

closed 2017-08-31 by ketai