ketai/ketai #16: bug when Instantiate our KetaiGesture object gesture?

Adiou, 2015-10-03:
Hi, 

First thank for this librairie which looks really great, but I can't make it work.

I'm working with APDE (https://play.google.com/store/apps/details?id=com.calsignlabs.apde&hl=en) and when I try to run your example, as soon as it's compiled, it close...
With the gesture example, I have : java.lang.noclassdeffounderror failed resolution of : Lketai/ui/KetaiGesture...

I really tried different way to install this library, but still not working...

Any clue ?

Thanks for your help!

A.

IngenieroElectronico, 2016-06-09:
Hello, I have Ketai_v9 and it works well however Ketai_v12 does not, I think the newer versions of Ketai does not work in APDE, look in this channel good information of APDE, Ketai and Arduino:

https://www.youtube.com/channel/UCyWRJgfJMu3fiW4xO0FmB1w/videos?sort=dd&shelf_id=0&view=0

ketai, 2016-07-04:
Try updating to the latest version of Android mode.  The latest android mode resolves a number of library issues that broke in the last version of the android mode.

closed 2016-07-04 by ketai