ketai/ketai #17: No virtual method getActivity()...

Adiou, 2015-10-15:
Hi, 
I tried all the examples from the Ketai library, and FaceDetector is the only one working...The other ones stop and I have this error message : "java.lang.NoSuchMethodError: No virtual method getActivity()Landroid /app/Activity; in class Lprocessing/core/PApplet; or its super classes (declaration of 'processing.core.PApplet' appears in /data/app/processing.test.gytoscope-2/base.apk"
Any clue about what's going on?
Thans for your help!
A.

Adiou, 2015-10-15:
All the problem come from (this), when we do stuff like "...=new KetaiSensor(this);"
Any advice how to fix it?
Cheers!
A.

Adiou, 2015-10-15:
I manualy reinstall the librairie (v12), and now I have
"java.lang.NoClassDefFoundError: Failed resolution of : Lketai/ui/KetaiGesture;"

manashmndl, 2016-04-26:
I am facing the same problem

IngenieroElectronico, 2016-06-09:
Hello, I have Ketai_v9 and it works well however Ketai_v12 does not, I think the newer versions of Ketai does not work in APDE, look in this channel good information of APDE, Ketai and Arduino:

https://www.youtube.com/channel/UCyWRJgfJMu3fiW4xO0FmB1w/videos?sort=dd&shelf_id=0&view=0

tmigone, 2016-06-21:
Hi, i'm also facing the same problem. Unfortunately i tried installing Ketai v9 with APDE but it seems like v9 does not get on the library list after being installed and i can't get that to work either.

IngenieroElectronico, 2016-06-21:
APDE Forum:

https://forum.processing.org/two/discussion/2342/android-pde-create-and-run-sketches-on-an-android-device

IngenieroElectronico, 2016-06-21:
I uninstall and reinstall Ketai_v9 in APDE and again, no problems. I have test it in Android 4.4.4, Android 5.0 and Android 6.0

ketai, 2016-07-04:
The latest version of android mode (v 3.0 as reported in the modes panel) resolves the issue as it has been restructured in a manner that is closer to version 1.0 (when ketai was initially created).  Version 2.0 of android mode broke ketai since PApplet was changed to extend a Fragment.  In version 3.0 of android mode PApplet now extends Object and has the applications Activity.

closed 2016-07-04 by ketai