ketai/ketai #19: Compiling error Ketai

imartini, 2015-11-03:
Hi!

I'm trying to compile Ketai from eclipse, but can not. I get a warning on the compilation ant:

[javac] Compiling 27 source files to C:\Users\Rider\workspace\test\ketai-master\bin
    [javac] C:\Users\Rider\workspace\test\ketai-master\src\ketai\camera\KetaiCamera.java:15: error: package javax.microedition.khronos.opengles does not exist
    [javac] import javax.microedition.khronos.opengles.GL10;
    [javac]                                           ^
    [javac] C:\Users\Rider\workspace\test\ketai-master\src\ketai\camera\KetaiCamera.java:19: error: package android.content.pm does not exist
    [javac] import android.content.pm.ApplicationInfo;
    [javac]                          ^
    [javac] C:\Users\Rider\workspace\test\ketai-master\src\ketai\camera\KetaiCamera.java:20: error: package android.content.pm does not exist
    [javac] import android.content.pm.PackageManager;
.
.
.

And 73 more errors. I need import some Android jar library? 

Thanks

IngenieroElectronico, 2016-06-09:
Look in this channel good information of APDE, Ketai and Arduino:

https://www.youtube.com/channel/UCyWRJgfJMu3fiW4xO0FmB1w/videos?sort=dd&shelf_id=0&view=0

ketai, 2016-07-04:
Make sure that there is a version of the ADK is associated with the project.  

Select the project and File->Properties will open the properties window.  Select Android and check a version of the ADK to compile against.  

[image]

closed 2016-07-04 by ketai