ketai/ketai #22: `KetaiBluetooth.getConnectedDeviceNames` produces both the name and the address of each device (rather than just the name)

mitchmindtree, 2015-12-11:
I ran into the issue when trying to check if a name I was looking for was in `KetaiBluetooth.getConnectedDeviceNames` using a string comparison.

I was anticipating that the method would behave just like the `KetaiBluetooth.getDiscoveredDeviceNames` and `KetaiBluetooth.getPairedDeviceNames` methods, who only produce the device names as their method names suggest.

Specifically, the `getDiscoveredDeviceNames` and `getPairedDeviceNames` produce names like:
`Foo`
`Bar`

whereas `getConnectedDeviceNames` produces names like:
`Foo(1F:5E:9A:32:1F)`
`Bar(3F:16:8A:3F:1E)`

You can see the line that performs the concatenation [here](https://github.com/ketai/ketai/blob/d7ff0994868f5f1f7a52221d8b9695f46fec2fc1/src/ketai/net/bluetooth/KetaiBluetooth.java#L279).

I'm able to get around this pretty easily by retrieving a substring, but just thought I'd leave an issue as this seemed a little oddly inconsistent.

Anyways, thanks a lot for the library, I've found it very easy to use otherwise :)

IngenieroElectronico, 2016-06-08:
Look
https://www.youtube.com/channel/UCyWRJgfJMu3fiW4xO0FmB1w/videos?view=0&shelf_id=0&sort=dd

ketai, 2017-08-31:
The KetaiBluetooth class has been refactored to address this.  

getConnectedDeviceAddresses() // HW addresses
getConnectedDeviceNames()      // BT Names
getConnectedDeviceLabels()      // Convenience method for Both (for display purposes)

closed 2017-08-31 by ketai