ketai/ketai #30: Unable to Use Ketai Gestures Example

timkaboya, 2016-04-22:
Processing Error says 
Could not find the image toci.jpg.
FATAL EXCEPTION: Animation Thread

Where is this toci.jpg located? 
How do I get this example working. 

Also what gesture does it track. The Example does not say this in plain English.

ketai, 2016-07-04:
toci.jpg is located in the data folder:  https://github.com/ketai/ketai/tree/master/examples/Gestures/data

What version of android were you using?  What version of processing? What version of android mode?

The reference states the following supported gestures (http://ketai.org/reference/ui/ketaigesture/): 

void onTap(float x, float y) - x, y location of the tap

void onDoubleTap(float x, float y) - x,y location of double tap

void onFlick(float x, float y, float px, float py, float v) - x,y where flick ended, px,py - where flick began, 
v - velocity of flick in pixels/sec

void onLongPress(float x, float y) - x, y position of long press

void onPinch(float x, float y, float r) - x,y of center, r is the distance change

void onRotate(float x, float y, float a) - x, y of center, a is the angle change in radians

closed 2016-07-04 by ketai

timkaboya, 2016-07-07:
This was a while back. 
I worked this out then. Just moved the joci.jpg to the correct directory. Thanks.