ketai/ketai #32: Ketai-Bluetooth is crashing App

chrmoeller, 2016-06-08:
`Hi` there, 
I tried to write an App, which should be able to control a Bluetooth-Shield. This is my code (the core):

```
import android.os.Bundle;

import ketai.net.bluetooth.*;
import ketai.sensors.*;

KetaiBluetooth bt;
KetaiSensor sensor;

void onCreate(Bundle savedInstanceState) {
  super.onCreate(savedInstanceState); 
  bt = new KetaiBluetooth(this);  // !!!!!!! <<<<<<<< HERE IS THE FAILURE!
}

void setup()
{
  bt.start(); // BT Service starten.
  bt.connectDevice("00:13:01:06:20:83"); // HC-05 BT-Modul "SIA-Flug-2"
  delay(2000);
  size(800, 480);

  sensor = new KetaiSensor(this);
  sensor.start();
  frameRate(20);
}

void draw()
{
  background(255, 255, 0);

}
```

When I delete the above marked line "bt = new KetaiBluetooth(this);" (and of course every depending line), the code runs. However when adding only the mentioned line, the App will crash right after being started. What has to be done?
Using Processing 3.1.1 on a Windows-10-64-Machine, SDK-API-Level 15, JDK 8u92, connected and recognized Samsung Galaxy S4.

Thanks for any advice!

IngenieroElectronico, 2016-06-08:
Check the sketch permisions:

bluetooth
bluetooth admin

Maybe You can be interested in:

https://www.youtube.com/channel/UCyWRJgfJMu3fiW4xO0FmB1w/videos?view=0&shelf_id=0&sort=dd

In these channel there are projects with ketai bluetooth and more

ketai, 2016-07-04:
What version of android mode are you using? 

Your code currently runs using Android Mode 3.0.  Please update your android mode and test again.

closed 2016-07-04 by ketai