ketai/ketai #35: Camera image upside down

xuv, 2016-08-01:
Not a big issue, but on a Nexus 5X with Android 6.0.1, with the CameraGettingStarted example, the video is upside down.

AdoHaha, 2016-12-08:
same for me, nexus 5

wolfm2, 2018-02-01:
Hi xuv and AdoHaha,
This is likely an issue for a handful of devices that mount the orientation sensor in an uncommon way. A couple years back the Nexus 5X/6 had the same issue with the older camera API leaving it unfortunately for the app developer to fix. Check out the statement from the Google tech lead below. (from: Reddit)

Best,
Michael

#######################
Hi - Tech lead for Android's camera framework here.

Image sensors on a compliant Android device can be mounted in one of two ways. Most devices use one way, so a lot of camera apps have never been tested on devices that pick the other way.

Because of manufacturing reasons, we needed to mount the Nexus 5X main sensor in the less-common (reverse landscape) orientation - the wires from the sensor chip wouldn't have fit otherwise.

...

This is also why a number of apps showed upside down preview on the Nexus 6 front-facing camera last year - it's also in the less-common orientation. But far fewer apps support the front-facing camera than the back-facing one, so it was less of an issue than it is for Nexus 5X.

So if you do see apps with upside-down camera preview on Nexus 5X, please let the developers know about this - Android's developer relations folks are trying to reach out to apps as we find them, and LG is also helping out, but it might take a bit before everyone's up to speed on this issue.

closed 2018-02-01 by wolfm2

xuv, 2018-02-01:
@wolfm2 Oh, thanks for that explanation. Very interesting.