ketai/ketai #36: Help me. Can you make it through the (Processing)JAVA?

SeongJongKwak, 2016-10-10:
I want to know the information is as follows.
 1. WIFI List name
 2. WIFI Connect

I want to know the information
Can you make it through the (Processing)JAVA?
It's not that I want to create a smartphone application. PC is a program through processing program.

ketai, 2017-09-06:
Ketai is a library for android.  For PC question please post your questions to a processing forum or a java forum on stack overflow.  Try this https://stackoverflow.com/questions/5378103/finding-ssid-of-a-wireless-network-with-java

closed 2017-09-06 by ketai

SeongJongKwak, 2017-09-07:
Dear Ketai,

 

Nice to meet you. ^^ 

 

I am using your library well.

And I respect you for your ability.

 

I have one favor.

I want you to update the library of your processing.

There is something wrong with the latest version of Android.

 

thank you.