ketai/ketai #42: change bluetooth read refresh rate?

al3k87, 2017-02-17:
So it seems like the string update of the bluetooth event when sending some values to the application is limit to around 100ms/10Hz, then it will start to accumulate the values received in the string, which will result in unfinished values added to the string from the data bluetooth event.
Is there a way to increase the bluetooth read speed?

What I want to achieve is sending 2 sensor int values which has a number from 0-250 and 0-2000, they are separated with a comma "," from the Arduino to the android processing application at a rate of at least 30Hz, but 60Hz++ is preferred.
Ive been searching around the web and others have found the same issues and it was pointed out to be in a source code where there was a thread sleep for 100ms, I dont know how to change this and build source code, therefore Im asking if there is a another way to deal with this or if its possible to make a option in the library code to change this in the sketch in future versions?

ketai, 2017-02-20:
We will be updating at the end of the month.  We will be looking at modifying these timers.

al3k87, 2017-04-04:
Any news on the update?

Flying-Bolt, 2017-06-01:
What´s about the update?

ketai, 2017-08-31:
Lowered the thread hold off to 5ms.

closed 2017-08-31 by ketai

al3k87, 2017-08-31:
Thank you, I think that should work good!
Have you also updated the compiled version in the download section?

ketai, 2017-08-31:
We are currently testing a final build and should be posted by next week at the latest.

al3k87, 2017-08-31:
Thanks for the fast response.
I know this is probably not the place to ask, and Im not gonna bother you
with further questions milking your time away, but if I want to compile the
source code myself, where can I find the info to do that, any pointers?

Thanks
Regards Alexader Nilsen