ketai/ketai #45: Camera crashing in Android 7.1.2

koser, 2017-03-30:
Dug into a bugreport captured by adb and found this snippet, hoping that helps find a fix:

--------- beginning of crash
03-30 09:55:59.585 30334 30354 E AndroidRuntime: FATAL EXCEPTION: Animation Thread
03-30 09:55:59.585 30334 30354 E AndroidRuntime: Process: processing.test.cameragettingstarted, PID: 30334
03-30 09:55:59.585 30334 30354 E AndroidRuntime: java.lang.NoSuchMethodError: No virtual method getActivity()Landroid/app/Activity; in class Lprocessing/core/PApplet; or its super classes (declaration of 'processing.core.PApplet' appears in /data/app/processing.test.cameragettingstarted-2/base.apk)
03-30 09:55:59.585 30334 30354 E AndroidRuntime: 	at ketai.camera.KetaiCamera.<init>(KetaiCamera.java:130)
03-30 09:55:59.585 30334 30354 E AndroidRuntime: 	at processing.test.cameragettingstarted.CameraGettingStarted.setup(CameraGettingStarted.java:29)
03-30 09:55:59.585 30334 30354 E AndroidRuntime: 	at processing.core.PApplet.handleDraw(Unknown Source)
03-30 09:55:59.585 30334 30354 E AndroidRuntime: 	at processing.core.PSurfaceNone.callDraw(Unknown Source)
03-30 09:55:59.585 30334 30354 E AndroidRuntime: 	at processing.core.PSurfaceNone$AnimationThread.run(Unknown Source)
03-30 09:55:59.588  1138  2947 W ActivityManager:   Force finishing activity processing.test.cameragettingstarted/.MainActivity

koser, 2017-03-30:
A bit more info:
device: Google Pixel
Processing 3.3, Android API version 25

ketai, 2017-08-31:
Issue was changes in latest android mode.  Code was updated for latest 4.0 rc3 android mode

closed 2017-08-31 by ketai