ketai/ketai #47: Bluetooth: unpaired devices not discoverable

cbernett, 2017-05-11:
Hello,

I'm using the bluetooth library with Arduino and it was working fine and then stopped working. Nothing significant was changed. So, I went back to your example to do a basic test and that is failing. I've narrowed it down to this problem: non-paired devices are not being discovered. If something is already paired to the phone, it sees it and shows me the option to pair it (again?) in the list. 

I am using a Nexus 5x with Android 6.0 currently. I was using it with 7.0 but tried reverting to 6.0 to see if that helps. It does not. I have SDK 6.0 (23) selected in processing and permissions are set properly. This issue was replicated on a Google Pixel with Android 7.1.2.

I have an urgent need for this - any help would be greatly appreciated!!

wolfm2, 2018-02-01:
Hi cbernett,
  If you are still having this issue, let me know.  Specifically you say non-paired devices are not being discovered but it worked fine the first time when (I'm assuming) your devices were not paired.  Is that correct?

Best,
Michael

usmanaruz, 2020-05-27:
> Hi cbernett,
> If you are still having this issue, let me know. Specifically you say non-paired devices are not being discovered but it worked fine the first time when (I'm assuming) your devices were not paired. Is that correct?
> 
> Best,
> Michael

As you said, I am facing the same problem

kola4k, 2021-12-28:
**+ same problem**