ketai/ketai #53: Bluetooth Problem!

VlogDev, 2017-11-18:
Hello,

I have the problem that he does not find the class bundle and intent.
Can who help me?

Best regards

[image]

wolfm2, 2018-02-01:
HI VlogDev,
  I'm happy to help.  It looks like you are running the example at http://ketai-scripts-ketai.c9.io/examples/bluetoothcursors/ .  Is this correct?

  Please copy and paste the full text of the error and the type of device (make and model) you were trying to use this on.

Best,
Michael

BarthelemyPassinCauneau, 2018-03-29:
I have the same problem on processing : 

No library found for android.content
No library found for android.os

I don't know how to import android's library, can you help me (us) ?