ketai/ketai #75: How to install ketai library in APDE?

bigboss97, 2019-11-20:
I've tried following:
Downloaded ketai.zip.
Open APDE on my phone, Tools -> Manage Libraries -> Install Compressed Library. Invalid file!

I also read somewhere saying to extract the files to the IDE library directory. Unfortunately, I couldn't find the directory.

EmmanuelPil, 2019-11-28:
You can unzip [this](https://drive.google.com/file/d/1QXmG6y6_B4IDYYknBeAd0iGdy2QUe6wQ/view?usp=drivesdk) in the sketchbook/libraries folder.

bigboss97, 2019-11-29:
Thanks for the reply.
Now I can see the library is installed. Then I tried the sample BluetoothCursors. APDE asked me to install preview and I confirmed.
[image]
But then I'm getting following error:
Build failed, check the console for more information.
------------
10 Problems (8 errors, 2 warnings)
Compiling with ECJ failed

I'm new in APDE and don't know how to open the console.

bigboss97, 2019-11-29:
Thanks for the reply.
Now I can see the library is installed. Then I tried the sample BluetoothCursors. APDE asked me to install preview and I confirmed.
But then I'm getting following error:
[image]

I'm new in APDE and don't know how to open the console in APDE.

EmmanuelPil, 2019-11-29:
You have to import the  oscP5  lib for that example.
Just to save you some time; unzip [this file](https://drive.google.com/file/d/1DF12h78E7BXxiim7fi2nsEYwsOiPaJO9/view?usp=drivesdk) in the libraries folder, like you did with the other one.
What is your need for the Ketai lib?

bigboss97, 2019-12-01:
Now I got rid of the errors, thanks. But APDE crashes with:
APDE Sketch Preview keeps stopping.
I tried different sample codes in Ketai and oscP5 and all ended up with the same  message. I'm using Android 8.

I want to do something with Bluetooth. I read that I  need Ketai to do that, right?

EmmanuelPil, 2019-12-02:
Does it crash I'm app mode also?
Go to settings, choose build/verbose output, then run the sketch. If it crashes select all the error text in the cinsole,  and post it here as if it were code.
Yes, I use ketai for bluetooth; working well.
Also; take a free hour and read this wiki carefully at ease. APDE is worth it.

bigboss97, 2019-12-02:
This is the console output of running in app mode:
[sketch.log](https://github.com/ketai/ketai/files/3909608/sketch.log)

EmmanuelPil, 2019-12-03:
The log doen't show any error.
How does it crash?
Do you see any screen.
If there is any error it probably is from the oscP5 lib. There are some issues.
Ketai Bluetooth is working well, and you don't need much code.
For instance if you want to connect to a bluetooth arduino shield like HC-05
You can make direct connection with: 
bt.getPairedDeviceNames();
bt.connectToDeviceByName("HC-05");

bigboss97, 2019-12-04:
In Preview mode, clicked on the Play button. Then a lot of text scrolling through console. The screen flashed once and a popup showed:
"APDE Sketch Preview has stopped"

In App mode, clicked on the Play button. Then a lot of text scrolling through console. It showed:
"Do you want to install an update to this existing application? Your existing data will not be lost. It does not require any special access"
Clicked on Install and the progress bar is shown. At the end a popup showed:
"BluetoothCursors has stopped."

Not sure what was updating. Play Store doesn't show any update for ADPE is required.

EmmanuelPil, 2019-12-04:
It is updating because the app already was installed previously. 
Ping me at the processing forum with @noel  I'll help you further there.

bigboss97, 2019-12-05:
https://discourse.processing.org/t/ping-noel-apde-crashes-on-running-sketch/16129

ToSStudio, 2022-06-02:
same problem here : APDE on a Samsung TAB4. I have copied the Ketai.zip file to several folders and extracted the ZIP there, too.
The library-installer cannot find them or crashes alltogether ("APDE has stopped").
The best (yet still not helpful) Msg I get is "The requested file does not exist. Try a different file manager".
Any hint would be appreciated