ketai/ketai #79: add a release to fix Processing PDE broken download

jeremydouglass, 2020-03-26:
Currently ketai is not listed in Processing Contributions Manager.

This is because these files listed in sources.conf:

http://ketailibrary.org/ketai.txt
http://ketailibrary.org/ketai.zip

...are no longer available since ketailibrary.org was decommissioned.

You can fix this very quickly in either of two ways:

1. add the ketai.txt and ketai.zip files to a path on ketai.org, and let us know what path.
2. upload the ketai.txt and ketai.zip files as attachments to a release on the Releases tab of this github repo! Then they can be linked from PDE and made automatically available to people who use Processing again.

with best regards,
Jeremy

processing/processing-docs#760 (comment)

ketai, 2020-03-27:
Hi Jeremy,

Thanks for catching this. We are on it (adding a release to Github). 
We‘ll fix the txt and zip on ketai.org as well.

Best,
Daniel

jeremydouglass, 2020-03-27:
Thank you so much, Daniel!

Any release that you make will automatically receive a dynamic tag "latest" through github -- you don't tag "latest" it manually, just a normal tag like "v1.2". The latest release will then have a direct url like this:

https://github.com/ketai/ketai/releases/download/latest/ketai.txt
https://github.com/ketai/ketai/releases/download/latest/ketai.zip

Just post which your preferred urls are for Contributions Manager: GitHub, or 

http://ketai.org/ketai.txt
http://ketai.org/ketai.zip

ketai, 2020-03-29:
> Links:

Hi Jeremy,

We've updated to use

  http://ketai.org/ketai.txt
  http://ketai.org/ketai.zip

Thank you,

/jesus

jeremydouglass, 2020-03-29:
@prisonerjohn -- could you update Contributions Manager with these new ketai urls please?

prisonerjohn, 2020-03-31:
Just updated it, thanks!

closed 2020-03-31 by jeremydouglass