ketai/ketai #83: Camera and orientation

jgmy, 2020-07-25:
HI! Under some circunstances, ketai camera crashes when the app has a fixed landscape layout but is started when the phone is in portrait orientation.

Look for the line with asterisks /******/

```
import ketai.sensors.*;
import ketai.camera.*;
import android.content.res.Resources;
import android.os.Build;
import java.util.Locale;

KetaiSensor sensor;
float compassX, compassY, compassZ, accuracy;

KetaiCamera cam;
PImage foto;
boolean camReady;

float cx, cy, radius;
void setup()
{
/********************************************/
/* Uncomment this and the sketch will fail */
/********************************************/
  //orientation(LANDSCAPE);
  fullScreen();  
  
  cx=width/2;
  cy=height/2;
  radius=max(cx, cy)/2;
  sensor = new KetaiSensor(this);
  sensor.start();
  

  textAlign(CENTER, CENTER);
  textSize(displayDensity * 36);
  cam = new KetaiCamera(this, min(width, 1024), min(height, 768), 24);
  foto=createImage(min(width, 1024), min(height, 768), ARGB);
  imageMode(CENTER);
  camReady=true;
  if (!hasPermission("android.permission.CAMERA") ) {
    camReady=false;
    requestPermission("android.permission.CAMERA", "PermissionRes");
  } else {
    cam.start();

    cam.autoSettings();
  }
}

void draw()
{
  float ladeo, inclinacion;

  background(78, 93, 75);
  if (camReady) {
    if (!cam.isStarted()) {
     // cam.start();
      // cam.autoSettings();
    } else {
      if (cam !=null && cam.isStarted() ) {
        image (cam, width/2, height/2, width, height);
      } 
    }
  }
  
  stroke(color(255, 255, 255));
  float in;
  for (float f=0; f<370; f+=90) {
    in=compassZ+f;
    strokeWeight(3);
    line(cx+cos(radians(in))*radius, 
      cy+sin(radians(in))*radius, 
      cx+cos(radians(in))*50, 
      cy+sin(radians(in))*50);
    stroke(color(0, 0, 0));
    strokeWeight(1);
    line(cx+cos(radians(in))*radius, 
      cy+sin(radians(in))*radius, 
      cx+cos(radians(in))*50, 
      cy+sin(radians(in))*50);
  }
  text("Compass: \n" +
    "Orientación: " + nfp(compassX, 1, 3) + "°\n" +
    "Inclinación: " + nfp(compassY, 1, 3) + "°\n" +
    "Ladeo: " + nfp(compassZ, 1, 3)+ "\n"+
    "accu: " + nfp(accuracy, 1, 3) + "\n" 
    , 0, 0, width, height);
}

void onOrientationEvent(float x, float y, float z, long tstamp, int accu)
{
  compassX = x;
  compassY = y;
  compassZ = z;
  accuracy= accu;
}

void onCameraPreviewEvent()
{ 

  cam.read();
}

void PermissionRes(boolean granted) {
  if (granted) {
    println("cam.start from PermissionRes");
    cam.start();
    cam.autoSettings();
    camReady=true;
  } else {
    println("No permission");
  }
}
```

JuzerNejm10, 2023-09-27:
You should add this line to your manifest under activity:
android:screenOrientation="landscape"

jgmy, 2023-09-27:
I will check it. Thanks.

JuzerNejm10, 2023-09-27:
I had the same problem. You're welcome.