ketai/ketai #87: Could you tell me how to read analog data from arduino using this library?

VigneshVelmurugan, 2021-01-05:
I have tried different solutions from the internet to achieve this using this library but I have only been able to go as far as establishing a connection. I have not been able to read the incoming data.

VigneshVelmurugan, 2021-01-05:
void onBluetoothDataEvent(String who, byte[] data) { 
  String Str = new String(data); 
  println(Str);
  
  //background(int(Str));
} 
Nothing gets printed on the console, even though arduino is transmitting.

EmmanuelPil, 2021-01-05:
Hi. Are you using Android PC or APDE?
Of course Bluetooth permissions are enabled?
What Android version are you using?

VigneshVelmurugan, 2021-01-05:
> Hi. Are you using Android PC or APDE?
> Of course Bluetooth permissions are enabled?
> What Android version are you using?

Hey,
 Thank you. I am using Android 10. I have enabled all Bluetooth permissions and am writing the code on the processing IDE on PC in android mode.

EmmanuelPil, 2021-01-05:
I believe (seeing this code), you are following [this topic](https://discourse.processing.org/t/android-bluetooth-to-arduino-and-vice-versa/26578/88?u=noel) on the forum. Would you like to enter the conversation there?
Anyway, one thing you can try is to give a large interval on the Arduino side, like delay(100); between each write() command.
Without seeing the whole code it's difficult to opinion, Did you use bt.start(); ?

VigneshVelmurugan, 2021-01-05:
Thank you for the help. Yea I did everything as above but I am now getting strange characters when I print in console.

My arduino code:
int index_angle;
void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
}

void loop() {
  // put your main code here, to run repeatedly:
  index_angle = map(analogRead(A0),0,1023,0,180);
  Serial.println(index_angle);
  delay(2);
}

Processing:
import ketai.net.bluetooth.*; 
import android.os.Bundle; 
//String device_name = "BT04-A"; 
 

KetaiBluetooth bt; 

void setup() { 
  fullScreen(); 
  orientation(PORTRAIT); 
  bt.getPairedDeviceNames(); 
  //bt.connectToDeviceByName(device_name);
  bt.connectDevice("00:14:03:05:EF:5D");
  bt.start();
} 

void draw() { 
  //background(int(Str));
} 

void onBluetoothDataEvent(String who, byte[] data) { 
  String Str;
  Str = new String(data);
  println("Value Received"+Str);
  //println(Str);
} 
void onCreate(Bundle savedInstanceState) { 
  super.onCreate(savedInstanceState); 
  bt = new KetaiBluetooth(this);
}

Console:
BUILD SUCCESSFUL in 38s
28 actionable tasks: 28 executed
Found onBluetoothDataEvent method.
BEGIN mConnectThread SocketType:Secure:DSD TECH HC-05
Found onBluetoothDataEvent method.
Socket Type: SecureBEGIN mAcceptThreadThread[Thread-4,5,main]
AcceptThreadSecure
KBTConnect thread connected!
create Connection thread to DSD TECH HC-05
BEGIN mConnectedThread to 00:14:03:05:EF:5D
Value Receivedï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½
Value Receivedï¿½
Value Receivedï¿½
Value Receivedï¿½
Value Receivedï¿½
Value Receivedï¿½
Value Receivedï¿½
Value Receivedï¿½
Value Receivedï¿½
Value Receivedï¿½
Value Receivedï¿½
Value Receivedï¿½

VigneshVelmurugan, 2021-01-05:
I also tried with delay(200) in arduino but got the same results.

VigneshVelmurugan, 2021-01-05:
I finally got it to work. Thank you. It worked after further increasing the delay to 1000

EmmanuelPil, 2021-01-05:
If an interval of 1 second is acceptable, then that's fine.
The data array string can vary much in length, depending on the speed the Arduino is sending. In part caused by Bluetooth’s channel frequency-hopping, it comes in chunks from a few bytes to hundreds of bytes at once. So you have to split the incoming string accordingly like: "for (String ts : Str.split("!")) {"  (Use "mySerial.print('!');" instead of "println();") Also you need to read the characters before the first '!' in the string and the characters after the last '!' character and dismiss them because the maybe only a part of the actual value. This way you will lose some bytes, but that not much, and you can use a low delay. Readings without a loss are also possible, but you'll need a more complex String handling.